Whilser/GyverLamp #2: Привет! было бы круто если бы ты смог добавить поддержку управления через мосфет, потому что уж очень удобная интеграция в hass но не хватает мосфета ) спасибо заранее)

Whilser, 2020-01-02:
Привет! было бы круто если бы ты смог добавить поддержку управления через мосфет, потому что уж очень удобная интеграция в hass но не хватает мосфета ) спасибо заранее)

_Originally posted by @shohart in https://github.com/Whilser/GyverLamp/issues/1#issuecomment-570073619_

Whilser, 2020-01-02:
Поясни для чего это нужно? Лента адресная, управляется через цифровой пин, после отключения энергию практически не потребляет. Для управления RGB лентами с интеграцией в  Home Assistant существуют конструкторы прошивкой, такие как ESPHome.io

shohart, 2020-01-04:
1. Как в том анекдоте «во-первых это красиво», из стремления к
перфекционизму и все такое. Почти не потребляет но все же определенное
потребление хоть и минимальное но есть.
2. Чтобы можно было использовать отдельный блок питания (при ленте на 12в)
в моем городе адресную ленту на 5в не купить, только заказывать с алика.
3. Чисто шкурный интерес: собрал себе лампу с мосфетом и прошивкой от
gunner’a но не смог разобраться с тамошним mqtt он там я так понял не
структурирован под hass. Хотел просто прошить но не ковырятся в мешанине
проводов.
4. Согласен есть и WLED прекраснейший и в esphome есть возможность, но в
wled вроде как нет поддержки матриц, а в esphome вроде все эффекты с
расчетом на плосскую матрицу а не на циллиндр как у нас, поюс там не все
эффекты есть из данного проекта и вообще ;))

Whilser, 2020-01-04:
Все же непонятно для чего использовать отдельный блок питания, на NodeMCU есть стабилизатор, питание с блока 12 вольт можно напрямую подавать на VIN, в случае с WEMOS D1 используйте внешний стабилизатор на 5 в. При выключенной ленте энергию потребляет только сам модуль - порядка 175 мА, это очень незначительно, физически отключать ленту нет смысла. Накидай схемку твоего подключения, пока мосфет в ней видится лишним.

shohart, 2020-01-06:
Схема простая - мосфет управляет питанием матрицы/ленты, nodemcu питается
от отдельного блока.
Но в принципе не суть в общем. Оказывается добрые люди уже написали
отдельный custom component для hass. вот
https://github.com/AlexxIT/GyverLamp*.*
Так что мой вопрос можно считать решенным.

closed 2020-01-06 by Whilser