Whilser/GyverLamp #3: Не могу подключиться по MQTT к Home Assistant

cepxuo, 2020-02-02:
Добрый день!
Спасибо огромное за доработку для Home Assistant (HA)! Очень хотелось бы всё это связать.
У меня на HA стоит Mosquitto брокер из одноимённого аддона Hass.io. Включена поддержка анонимных пользователей, и в acl прописано чтение/запись для всех пользователей. Всякие мои железки на Sonoff подключаются и работают, а вот с лампой пока не получается. Влогах так:

```
1580670185: New connection from 192.168.хх.хх on port 1883.
[ERROR] Auth error with DEVS_USER
1580670188: Socket error on client <unknown>, disconnecting.
1580670195: New connection from 192.168.хх.хх on port 1883.
1580670195: Socket error on client <unknown>, disconnecting.
1580670279: New connection from 192.168.хх.хх on port 1883.
1580670279: Socket error on client <unknown>, disconnecting.
1580670289: New connection from 192.168.хх.хх on port 1883.
1580670289: Socket error on client <unknown>, disconnecting.
1580670309: New connection from 192.168.хх.хх on port 1883.
1580670309: Socket error on client <unknown>, disconnecting.
1580670349: New connection from 192.168.хх.хх on port 1883.
1580670349: Socket error on client <unknown>, disconnecting.
1580670429: New connection from 192.168.хх.хх on port 1883.
1580670429: Socket error on client <unknown>, disconnecting.
1580670589: New connection from 192.168.хх.хх on port 1883.
[ERROR] Auth error with DEVS_USER
1580670591: Socket error on client <unknown>, disconnecting.
```
Пробовал и с разными пользователями и с паролями и без, и совсем без пользователей, и с пользователем anonymous - всё также в логах.

Подскажите, пожалуйста, где что можно подкрутить?

Whilser, 2020-02-06:
Добрый день! Какой именно аддон установлен? Судя по логу, не проходит аутентификация. Имя пользователя и пароль задается при первоначальной настройке лампы. У меня установлен Mosquitto брокер MQTT.org с конфигурацией

```
logins:
  - username: USER
    password: PASSWORD
anonymous: false
customize:
  active: false
  folder: mosquitto
certfile: fullchain.pem
keyfile: privkey.pem
require_certificate: false
```
И еще, хотелось бы взглянуть на лог из консоли лампы.

cepxuo, 2020-02-09:
У меня тоже Mosquitto-broker. Конфиг такой:
```
logins: []
anonymous: true
customize:
  active: true
  folder: mosquitto
certfile: fullchain.pem
keyfile: privkey.pem
require_certificate: false
```
И в /config/mosquitto/accesscontrollist прописано так:
```
user []
topic readwrite #
```
При этом разные железки коннектятся и взаимодействуют с брокером. Конкретных юзеров не создавал, так как на железках генерится разный юзернейм.

sanyadez, 2020-09-05:
Поменяйте плату управления. Я изначально использовал node mcu, началось аналогочное. Повесил на wemos d1 mini, так же, перешил еще пару, завелось безотказно, тьфу-тьфу. Работает больше года. Хотя, на ноде тоже работало около того и начало отваливаться. Смотрите в логах, там в авторизацию непонятное слать начинают. Дело не в коде, а в плате

closed 2020-10-23 by Whilser