Whilser/GyverLamp #4: Настройка переключений в HA

DimsterTim, 2020-02-08:
Поделитесь пожалуйста настройкой автоматизаций в Home Assistant для переключения эффектов. 

Заметил, что лампа подвисает на несколько секунд - приостанавливается анимация лампы при обновлении веб страницы. Это неизбежно?

Whilser, 2020-02-09:
Вообще, не должно ничего подвисать при обновлении веб страницы. Покажи лог из консоли лампы во время обновления веб страницы. 

Пример автоматизации в ХА:

```

- alias: Gyver Lamp Effect
  trigger:

  action:
    - service: light.turn_on
      entity_id: light.gyver_lamp
      data:
        effect: Конфетти      
```

sashayohan, 2020-02-17:
У меня тоже заметил есть подвисания, например горит огонь, меняешь яркость и эффект притормаживается, потом после смены яркости дальше плавненько работает.

Fortunatto, 2020-04-13:
> У меня тоже заметил есть подвисания, например горит огонь, меняешь яркость и эффект притормаживается, потом после смены яркости дальше плавненько работает.

Подтверждаю. Аналогичная ситуация - при обновлении страницы в браузере или при переподключении к ней, текущий эффект замораживается на время, потом, после прогружки страницы браузера, эффект размораживается.

closed 2020-05-03 by Whilser