Whilser/GyverLamp #7: Error sending discover message

sergeolkhovik, 2020-03-15:
Большое спасибо за поддержку HA, но, похоже, столкнулся с проблемой discovery (в логах вижу Error sending discover message) и не могу решить её.
По конфигурации: прошивка 1.5, на стороне HA: Ubuntu, свежий HA (0.106.6), mosquitto.
Включил debug, раскомментировал вывод hass_discover_str, 

```
16:41:34.433 -> Attempting MQTT connection to mqtt.home on port 1883 as ...connected!
16:41:34.467 -> {"~":"homeassistant/light/ESP-47fbb1","name":"Gyver Lamp ESP-47fbb1","uniq_id":"47fbb1","avty_t":"~/state","pl_avail":"online","pl_not_avail":"offline","bri_cmd_t":"~/brightness/set","bri_stat_t":"~/brightness/status","bri_scl":255,"cmd_t":"~/switch","stat_t":"~/status","fx_cmd_t":"~/effect/set","fx_stat_t":"~/effect/status","rgb_cmd_t":"~/rgb/set","rgb_stat_t":"~/rgb/status", "device": {"ids": ["47fbb1"], "name": "Gyver Lamp", "mf": "Alex Gyver", "mdl": "Gyver Lamp v2", "sw": "1.5.5 MQTT"}, "fx_list": ["Конфетти", "Огонь", "Радуга верт.", "Радуга гориз.", "Смена цвета", "Безумие 3D", "Облака 3D", "Лава 3D", "Плазма 3D", "Радуга 3D", "Павлин 3D", "Зебра 3D", "Лес 3D", "Океан 3D", "Цвет", "Снегопад", "Матрица", "Светлячки",  "Аквариум", "Звездопад", "Пейнтбол", "Демо"] }
16:41:34.535 -> Error sending discover message
16:41:34.535 -> Error sending discover message
16:41:34.535 -> Error sending discover message
```

Плюс в настройках HA mqtt добавил (https://www.home-assistant.io/docs/mqtt/discovery/):
```
mqtt:
  broker: localhost
  client_id: HA
  discovery: true
  discovery_prefix: homeassistant
```

Всё выглядит красиво, но именно discovery не работает. При этом я добавил контролы input_number, могу менять значения и вижу, что скорость реально работает, эффекты работают медленнее.
Что может быть не так? Параметр retention = true?
Сделал вот так для проверки:
```
  #ifdef DEBUG
  Serial.println(hass_discover_str);
  mqttclient.publish(String("homeassistant/light/"+clientId+"/config").c_str(), "", true);
  #endif
```
И всё прекрасно проходит:
```
homeassistant/light/ESP-47fbb1/config (null)
```

Я смотрю по коду, там дальше куча таких же вызовов с retention = true, вижу такие сообщения:

```
homeassistant/sensor/ESP-47fbb1U/uptime 64
homeassistant/sensor/ESP-47fbb1W/WiFi/RSSI -36
homeassistant/sensor/ESP-47fbb1W/WiFi/RSSI_pct 128
homeassistant/sensor/ESP-47fbb1W/WiFi/channel 1
```

То есть не в этом параметре дело.
И не в самих данных, похоже, тоже, вот вручную отправил:

[image]

И я вижу эту лампу в HA:

[image]

Буду рад любой идее или подсказке.
Спасибо!

Whilser, 2020-03-15:
Привет, попробуй в библиотеке PubSubClient в файле PubSubClient.h выставить `#define MQTT_MAX_PACKET_SIZE 1024`

В ХА дополнительно не нужно никаких настроек MQTT указывать.

sergeolkhovik, 2020-03-15:
В GyverLamp_v1.5_MQTT.ino уже есть
```
#define MQTT_MAX_PACKET_SIZE 1024
```

Но я его перенёс выше, до включения
```
#include <PubSubClient.h>
```

В mqtt файле прописал вот так:
```
  #ifdef DEBUG
  Serial.printf("MQTT_MAX_PACKET_SIZE = %d\n", MQTT_MAX_PACKET_SIZE);
```
В выводе вижу те же 1024:
```
19:31:12.453 -> Attempting MQTT connection to mqtt.home on port 1883 as ...connected!
19:31:12.487 -> MQTT_MAX_PACKET_SIZE = 1024
```
Но это ничего не помогло. Более того, сделал даже просто вот так:
```
  #ifdef DEBUG
  Serial.printf("MQTT_MAX_PACKET_SIZE = %d\n", MQTT_MAX_PACKET_SIZE);
  hass_discover_str = "{\"~\":\"homeassistant/light/ESP-47fbb1\",\"name\":\"Gyver Lamp ESP-47fbb1\",\"uniq_id\":\"47fbb1\",\"avty_t\":\"~/state\"}";
```
и всё равно получаю Error sending discover message :(

По поводу HA, у них написано (https://www.home-assistant.io/docs/mqtt/discovery/), что по-умолчанию выключено:
```
discovery
(boolean)(Optional)
If the MQTT discovery should be enabled or not.
Default value: false
```

Whilser, 2020-03-15:
Нет, ты измени это **прямо в файлах библиотеки**. По непонятным причинам переменная не переопределяется пользователем, хотя директива стоит.

sergeolkhovik, 2020-03-15:
Да, похоже, именно в PubSubClient.h нужно зажать размер.
Заработало, спасибо большое!

sergeolkhovik, 2020-03-20:
Ещё один вопросик, не по теме, но не знаю, куда писать. Можно ли обновлять прошивку по воздуху (в случае выхода новой версии), и как это сделать? Какими средствами?

Whilser, 2020-03-20:
Да, можно. Для этого в Arduino IDE нужно выбрать "сетевой порт" и нажать кнопку "загрузить" как обычно при компиляции новой прошивки. Лампа обновится по воздуху.

Whilser, 2020-05-03:
Немного изменил способ отправки discover message, сейчас все должно работать без правки библиотек.

closed 2020-05-03 by Whilser