Whilser/GyverLamp #8: extra effect

cosmos1978, 2020-03-15:
This seems like an effect that can easily be integrated.

https://gist.github.com/kriegsman/5adca44e14ad025e6d3b

Whilser, 2020-03-16:
Hi! The main problem with this sketch is that it uses the SmartMatrix library for Teensy USB Development Board, but we usually use NodeMCU board, based on ESP8266 Chip on board.

cosmos1978, 2020-03-16:
Ok I see now.

But I did find this one

https://gist.github.com/anonymous/68298debac462330719b

uses only fastled library and looks realy nice.
http://www.youtube.com/watch?v=EC1MuRhXd1U

http://www.youtube.com/watch?v=0Ehj7sEwOy4

cosmos1978, 2020-03-16:
This guy has a huge collection

https://www.youtube.com/channel/UCkOEj4gOrUq1rbKElkmdo9g

cosmos1978, 2020-03-20:
https://github.com/StefanPetrick/FunkyClouds

Whilser, 2020-03-20:
Ok, thank you. I'll see later which of these effects will look good on the lamp. Some of them  I will add maybe when I have time for this.

closed 2020-10-23 by Whilser