Whilser/GyverLamp #9: Эффект мигалки

iswitch, 2020-03-16:
Добавьте пожалуйста эффект мигалки (проблесковый маячок). Чтоб полоска выбранного цвета бегала по кругу с заданной скоростью, ну и можно по краям полоски сделать рассеивание. Спасибо)

Whilser, 2020-05-05:
Добавил эффект "полицейская сирена" в последнем обновлении. Удобно для использовании в сценариях с предупреждениями. Масштабирование меняет стиль эффекта, скорость - скорость мерцания.

iswitch, 2020-05-06:
Спасибо)

sashayohan, 2020-05-06:
Добавьте, эффект, что бы она крутилась по кругу. Как на старых машинах.
По типу такой https://youtu.be/NYUObVYJtGA
Лампа больше такую форму мигалки напоминает

sashayohan, 2020-05-24:
Ну уже почти получилось) Только почему то скорость рывками при 0, а при 255 совсем медленно.
И два отдельных цвета реально сделать? (только красный или только синий, и оранжевый до кучи, чтобы не вместе)

Whilser, 2020-05-25:
Чтобы убрать один из огоньков нужно закомментировать строчку drawPixelXY(ppp5, y, CRGB::Blue); Здесь вместо CRGB::Blue можно поставить свой цвет какой хочешь из палитры. При скорости 0 возможно не успевает отрабатывать. Бегунок скорости здесь по сути задает интервал времени между каждой прорисовкой фрейма. По сути когда ноль - получается цикл перерисовки выполняется каждый раз при входе в цикл эффекта.

closed 2020-10-23 by Whilser