Whilser/GyverLamp #10: English translation

cosmos1978, 2020-03-17:
I did a full translation to English.
Are you interested in the code ?

cosmos1978, 2020-03-25:
The webserver part is also translated. 

ust let me know and I will email the code

Stokstaart, 2020-03-26:
> I did a full translation to English.
> Are you interested in the code ?

I am! Having some issues doing the translationsby myself!

cosmos1978, 2020-03-26:
I prefer to give it to whilser so he can continue with the english version which would target a much larger  group of electronics enthousiasts.

Stokstaart, 2020-03-26:
I totale onderstand. I have no intentions of publishing it, it would be just for personal use!

cosmos1978, 2020-04-20:
I forked the code with translation here

[https://github.com/cosmos1978/GyverLamp/tree/Translation]

vortigont, 2020-04-20:
@cosmos1978 I'm interested in publicly available english version, might involve more people in there. Now I'm working on code refactoring, how about try to merge in your translation?

cosmos1978, 2020-04-20:
Sure, I will give it a go this evening.
is it this repo ? https://github.com/vortigont/GyverLamp/tree/featured

I will fork it and add the translation. But perhaps it is better to wait until your work is done and Whilser includes it in his master branch ?

Whilser, 2020-04-20:
Ok, thanks for the work. I'll consider best way to integrate localization.

Whilser, 2020-04-20:
I added English localization to the firmware. To switch to English localization, just uncomment the ENG directive.

vortigont, 2020-04-21:
@cosmos1978 it is, testers/bug reports are welcome :)
it's gonna take a while to clean up all messy places, lot's of ideas to implement.

closed 2020-05-03 by Whilser