Whilser/GyverLamp #13: Самостоятельное включение и перезагрузка лампы

Fortunatto, 2020-04-13:
Интеграция с HA. Заметил 3 особенности работы лампы:
1) Если по каким-то причинам сервер HA перезагружается и лампа в этот момент была выключена, она включается самостоятельно - не замечал, пока жена не спросила "Ты специально лампу на ночь оставляешь?"... Я это поведение лампы даже как датчик перезагрузок HA использовал, когда стал подозревать что карта памяти НА умирает. Возможно лампа не может достучаться до mqtt-сервера и срабатывает перезагрузка, после которой она включается на крайнем работавшем эффекте.
2) С определенной периодичностью происходит такая цепочка: лампа работает -> эффект "подвисает" -> происходит отключение всех эффектов -> вспыхивают белым несколько крайних светодиодов -> эффект запускается заново. Происходит при стабильно работающем НА и mqtt. Создается впечатление, что с течением времени происходит переполнение буфера(?), после чего лампа перезапускается.
3) Если НА и mqtt не работают, то лампа или совсем не реагирует на сенсор или реагирует начиная со второго прикосновения.

vortigont, 2020-04-14:
по 1) - так и есть, в процедуре реконнекта к MQTT есть код, который перегружает ESP после 10 неудачных попыток подключения. Оправданно ли такое поведение - большой вопрос, я бы убрал.
по 2) - надо дебажить
по 3) - в коде полно блокировок, нет асинхронности. В обработчике кнопки вызывается процедура обновления статуса MQTT, который мертв :)

Whilser, 2020-04-14:
Касательно первого вопроса - это происходит когда лампа некоторое время не может достучаться до MQTT сервера. Тогда она считает, что код, отвечающий за работу с MQTT завис и выполняет перезагрузку. Такое встречалось у некоторых пользователей, поэтому была добавлена данная функция. Хотя все же немного странно, статус лампы должен сохраняться после перезагрузок. Если используешь актуальную версию прошивки и лампа включается самостоятельно если она была выключена - это ошибка с которой нужно разобраться. По второму вопросу - такое происходит когда модуль крашится. Из-за чего это происходит пока непонятно, но отмечается некая корреляция с уровнем сигнала WiFi. По третьему вопросу, вероятно это связано с задержкой при попытке восстановить соединение с MQTT сервером. Если он. не используется - его можно отключить, тогда задержек не будет. В нормальном состоянии MQTT должен большую часть времени быть доступен. все это в принципе можно поправить если проблема будем масштабной и актуальной для большого числа пользователей.

sashayohan, 2020-04-17:
Тоже несколько раз лампа отваливалась и не могла подключиться к MQTT.
Два раза даже приходилось полностью перепрошивать, заново водить пароли и настраивать. Никакие перезагрузки не помогали.
Так же несколько раз замечал что лампа светилась, но в HA она отображалась как выключена, нажимаешь на выключить и включить, тогда статус уже начинал опять меняться.
Повторяется на всех трёх лампах.

sergeolkhovik, 2020-04-17:
У меня тоже были проблемы с отображением состояния в HA, и ещё в режиме
огня масштаб не менял цвет (постоянно! Через родное приложение без
проблем), смогу чуть позже заняться отладкой и предоставить больше
информации, хочу корпус допечатать.

vortigont, 2020-04-18:
Я исправил целую пачку проблем с mqtt/UDP и полностью переписал планировщик.
Если интересно, попробуйте код из моей тестовой ветки - [vortigont/GyverLamp/tree/featured](https://github.com/vortigont/GyverLamp/tree/featured)

vortigont, 2020-04-18:
@sashayohan  >Так же несколько раз замечал что лампа светилась, но в HA она отображалась как выключена, нажимаешь на выключить и включить, тогда статус уже начинал опять меняться.

Эту проблему тоже словил, исправлена в последнем комите

sashayohan, 2020-04-20:
> Я исправил целую пачку проблем с mqtt/UDP и полностью переписал планировщик.
> Если интересно, попробуйте код из моей тестовой ветки - [vortigont/GyverLamp/tree/featured](https://github.com/vortigont/GyverLamp/tree/featured)

Да надо поставить PlatformIO, разобраться как через него прошивать.

sashayohan, 2020-04-20:
> Я исправил целую пачку проблем с mqtt/UDP и полностью переписал планировщик.
> Если интересно, попробуйте код из моей тестовой ветки - [vortigont/GyverLamp/tree/featured](https://github.com/vortigont/GyverLamp/tree/featured)

Несколько вопросов по platformio.
1. Как поменять размер матрицы, строки закомментированы, а где флаги сборки с этими параметрами я так и не смог найти.
2. Ну и как собрать прошивку с командой 'pio run' так и не понял, где эту команду писать. 
3. Можно ли обновиться по воздуху? если да, то тоже подскажите куда тыкать.

Как будет время, напишите для таких как я, небольшой readme

vortigont, 2020-04-21:
@sashayohan можно собирать в Arduino IDE если так привычнее, platformio не обязателен,
небольшой ридми добавлю чуть позже.
1. Да просто раскомментируйте их. Файл config.h.default нужно скопировать под именем config.h. Эту будет ваш конфиг, который не будет меняться при обновлениях репозитория.
3. По воздуху обновлять можно, добавлено в последние изменения вилсера, просто открываете в браузере http://esp-xxxx/update и там через форму заливаете новый файл прошивки

vortigont, 2020-04-22:
@sashayohan накидал краткую [инструкцию](https://github.com/vortigont/GyverLamp/blob/featured/platformio.md), и [список изменений](https://github.com/vortigont/GyverLamp/blob/featured/gyverlamp/CHANGELOG.md)

Whilser, 2020-05-03:
Кажется нашел причину того, почему после перезагрузки esp лампа не восстанавливала свой статус что могло проявляться как самостоятельное включение лампы, при этом в HA она отображалась как выключенная. В последнем обновлении поправил.

ya-erm, 2020-05-04:
@vortigont 
> Я исправил целую пачку проблем с mqtt/UDP и полностью переписал планировщик.
> Если интересно, попробуйте код из моей тестовой ветки - [vortigont/GyverLamp/tree/featured](https://github.com/vortigont/GyverLamp/tree/featured)

Не сразу удалось скомпилировать код из ветки featured.
Ошибки компиляции, с которыми я столкнулся:
`gyverlamp\gyverlamp.h:25:18: fatal error: TZ.h: No such file or directory`
`gyverlamp/gyverlamp.ino:323:31: error: invalid conversion from 'const char*' to 'int'`
`gyverlamp/gyverlamp.ino:323:31: error: too few arguments to function 'void configTime(long int, int, const char*, const char*, const char*)'`
`gyverlamp/time.ino:54:88: error: 'DAWN_MAX_BRIGHT' was not declared in this scope`

closed 2020-10-23 by Whilser