Whilser/GyverLamp #16: New effect pendulul

cosmos1978, 2020-04-23:
really nice effect. I also added the possibility to change the colorpalette with the scale control.

You have 2 open pull requests :-)

Whilser, 2020-04-23:
ok, I will see as soon as I have free time. Pull requests are best done not by completely replacing files, but by making changes to sections of code. It’s easier to edit and make changes.

cosmos1978, 2020-04-23:
shouldn't a merge be automatically ?
But I will remember the next time I add some code :-)

cosmos1978, 2020-04-26:
I added a new effect ( patternwave ) that I made myself using 2 sine functions. fully transformable using scale.
I also used git to update the changes :-) so no more full files being updated.

Whilser, 2020-05-03:
Finally, I added  warm light and a pendulum effects. I added manually because Initially pull request was made without using git. The wave effect seemed incomplete to me, so I did not add it for a now.

closed 2020-05-06 by Whilser