Whilser/GyverLamp #20: weather mode

cosmos1978, 2020-05-07:
I am working on a weather effect that bases its effect on the weather forecast via openweathermap.
It has a new tab in the webbrowser to select how far upfront the forecast must be made.
It can show effects for clear,cloudy,rain,thunderstorm and snow.

For the moment I only have a fixed color configured for each weather type. So if you know some cool effects that would fit for those please let me know.

I only need to code the routine so the settings in the webform are saved to eeprom when that is done it is complete.

Whilser, 2020-05-07:
Great idea!

I am working now on the web server code too, I am adding the ability to change the settings of the mqtt broker.

As weather effects, you can use the effects of a lamp - snow, clouds. You can change the matrix effect if you make the color of the lines white to simulate rain and make the effect of the sun (gradient from red to yellow to half the lamp). It is also a good idea to add the temperature output to the lamp during the effect.

cosmos1978, 2020-05-07:
ok saving to eeprom is done :-)
Now mapping some cool effects to it.

cosmos1978, 2020-05-07:
Its all done. However adding the temperature as overlay on top of the active effects seems difficult.
I need to think about that.
I did a few custom effects and some addaptations to the snowstormeffect so it shows only droplets with shades of blue. ( light blue to dark blue ).
I also added a special random lightning effect that runs on top of the rain effect whenever lighnting is forecasted :-)

cosmos1978, 2020-05-07:
I pushed it to my github test branch. Feel free to test.

cosmos1978, 2020-05-08:
Got stuck now. The json response from openweathermap is to large to fit in memory.
So for now I can only get the forecast for the next 6 hours because I add an argument to only response with the next 6 hours. If I go larger I use to much memory.

I have read there are trick with PROGMEM to save in on flash instead of memory. Do you have any experience with that ? I am having issues to find some examples with HTTPclient.get and progmem.

cosmos1978, 2020-05-11:
Did some testing with arduinojson but it just wont work unless we go to an esp32 I gues its out of the question.

So for now the forecast is limited to +6 and +12 hours. This works perfectly

sergeolkhovik, 2020-05-12:
Interesting, is it possible to simply read some data from MQTT as it already has such connection? In this case I'd better retrasmit some data from HA (DarkSky or OWM) instead of doing this work on such a small device.

Whilser, 2020-05-12:
I agree with that. This is still an interesting project for testing, even a separate project. But I believe that it does not match the lamp concept. It takes up a lot of resources on small devices, which may limit further development. I think weather indication is best from home automation system through automation if it necessary.

cosmos1978, 2020-05-12:
I don't fully agree as not everyone is using homeassistant but still still would like to use such a feature. And the resource impact is not that high when using char arrays. For now I still use a string to capture the json payload. But I am now testing it with removing strings altogether and only using small chunks of data using the getStream function as explained here https://arduinojson.org/v6/how-to/use-arduinojson-with-esp8266httpclient/

When done correctly I see no impact on resources.

cosmos1978, 2020-05-12:
I just updated the code using getStreamto download in chunks and a specific deserialize filter so only the parts that I am interested are used. This uses almost no Ram memory :-)
Pretty elegant solution if I say so myself.

cosmos1978, 2020-05-12:
please have a look at my test repo on github.

cosmos1978, 2020-05-13:
You will need to add a openweathermap api key.

closed 2022-01-05 by Whilser