Whilser/GyverLamp #24: cannot connect to mqtt

alfadex, 2020-09-10:
indeed i am trying 2 days and i have not manage to connect to mqtt. I am using rpi 4 broker with HA

i get this error 

```
WiFi manager...connected! IP address: 192.168.1.235. Signal strength: 96%

MAC: A0:20:A6:02:0E:E0
UDP server on port 8888
Launched web server at: http://ESP-20ee0.local/
Attempting MQTT connection to ⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮?⸮\ @ on port ⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮?⸮\ @ as ⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮?⸮\ @...failed, rc=-2 try again in 10 seconds
Attempting MQTT connection to ⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮?⸮\ @ on port ⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮?⸮\ @ as ⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮?⸮\ @...
```
version 1.5
```
char mqtt_password[32] = "mqtt";
char mqtt_server[32] = "192.168.1.103";
char mqtt_user[32] = "mqtt";
char mqtt_port[10] = "1883";
byte mac[6];
```

Mosquitto broker
```
logins:
  - username: mqtt
    password: mqtt
anonymous: false
customize:
  active: false
  folder: mosquitto
certfile: fullchain.pem
keyfile: privkey.pem
require_certificate: false
```

Whilser, 2020-10-23:
Hi! Try upload the firmware with completely erasing all flash contents.

sergeich5, 2020-11-24:
Did the answer above solve the problem?

alfadex, 2020-12-27:
Yes that solve the problem, thank you. All work great, but i want only to mention i have many wifi disconnection. My power strengh is 92%. I see this from mqtt status, it say ofline and pattern stick until it will connect again. Is it because every minute it make reset? I am using esp8266.

Whilser, 2021-01-04:
>  i have many wifi disconnection

Try to remove `WiFi.setOutputPower(20);`

alfadex, 2021-02-07:
yes i think it is better now, thank you

closed 2022-01-05 by Whilser