Whilser/GyverLamp #28: Приложение

balbulator, 2020-12-08:
Скажите, а какое именно приложение для телефона надо использовать? Пробовал GyverLamp, FireLamp, wifiLamp. Ни одно не работает. Также возник вопрос, как включить будильник рассвет в Home assistant. В каком режиме (точка доступа или в локалке) можно воспользоваться  http://ESP-ID.local/.
Из общих вопросов: не планируете добавить функционал сенсорной кнопки как у Gunner47, удобно же?
Буду благодарен за Ваши ответы

Whilser, 2020-12-08:
Оригинальное приложение от Алекса вроде бы должно работать если он не изменил формат команд, я управление по UDP не корректировал. Вообще, с этой прошивкой можно управлять из веб интерфейса, нужно зайти по IP адресу лампы. IP адрес лампы можно по пятикратному быстрому нажатию на сенсорную кнопку узнать, ну или подсмотреть на роутере. Там же в веб интерфейсе можно настроить будильники. Подключаться нужно в штатном режиме, когда лампа подключена к сети Wi-Fi. 

Что за функционал сенсорной кнопки? Не слежу за версиями прошивок.

balbulator, 2020-12-08:
Подключился к домашней Wi-Fi сети (192.168.0.***). Лампа удачно подключилась после первичной настройки из ее точки доступа (192.168.*.***). Зайти из локальной сети (телефон или ПК, подключенные в локальной сети) невозможно ни по IP-адресу, который присвоен лампе (по 5-кратн. нажатию на сенсор), ни по  http://ESP-ID.local/. Оригинальное приложение от Алекса из плеймаркета GyverLamp не коннектится. 
Функционал кнопки сенсора:
1 клик - вкл/выкл
2 клика - след. эффект
3 клика - пред. эффект
удержание - яркость +/-
4 +4 клика - переход в режим прошивки ОТА
5 кликов - IP адрес
6 кликов - время
7 кликов - переход из локальной сети в точку доступа (например 192.168.4.1)
1 клик + удержание - скорость +/-
2 клик + удержание - масштаб +/- 
Это реально оставляет все возможности для ручного управления
Ваша прошивка мене очень понравилась из-за интеграции в Home Assistant, сейчас эксперементирую. Конечно карточка управления в ловелас убогая (проблема Home Assistant), но думаю под такую красоту можно и кастомную, намного функциональнее карточку оформить. Это мой следующий шаг. 
И пока не вижу возможностей в Home Assistant установить будильник "Рассвет"
Если есть возможность помочь с этими вопросами, буду благодарен
С уважением

Whilser, 2020-12-08:
А у тебя актуальная версия прошивки? Веб интерфейс позже добавился. Вместо ID  в `http://ESP-ID.local/.` нужно подставить ID чипа в шестнадцатеричном формате, но проще по IP адресу зайти, набрав http://192.168.0.*, все должно работать. Проверить приложение Алекса не могу т.к. андроида нет, возможно он там что-то поменял, но проще использовать веб интерфейс вместо приложения. 

По поводу кликов, первые 3 позиции +удержание на яркость и так реализованы, перевод в режим ОТА делать с кнопки не вижу смысла, он и без кнопки отлично работает, вывод IP адреса уже реализован. Но как показывает практика, уже даже 5 кликов делает неюзабельным использование сенсора, поэтому увеличивать количество кликов для управления не вижу смысла. Клик + удержание в принципе интересная идея, я подумаю, возможно реализую. 

А будет ли удобна установка будильника через MQTT? В ХА можно сделать автоматизацию установки времени будильника по условию через MQTT запросы.

balbulator, 2020-12-08:
Скачал и установил отсюда https://github.com/Whilser/GyverLamp прошивка GyverLamp_v1.5_MQTT.ino, сдается это самая последняя.
Вместо ID в http://ESP-ID.local/ подставил свой номерок, он и в MQTT тоже такой же, получилось http://ESP-b10b9f.local/, к примеру. По адресу в локалке http://192.168.0.** или http://192.168.0.**:8888 зайти также невозможно, думаю потому и проблемы с приложением на телефоне. А вот в Home Assistant все отлично работает, Home Assistant и MQTT на малинке Pi4. В прошивке от Алекса и от gunner47 все работало в приложениях на телефоне, но это не критично.

Возможно не весь функционал нужен на сенсоре, но все же лучше чем 5 раз шнур выдергивать, все это на Ваше усмотрение.

А вот как сделать в ХА будильник через MQTT? Мне очень интересно. Если я Вас не слишком утомил, то укажите как.

Пока писал это сообщение, одна проблема решилась. В роутере внутренний брадмаузер мутил воду, отключил его и заработало приложение GyverLamp, но все еще не хочет заходить по IP, хотя что-то начинало грузиться, пока был отключен Home Assistant. Может быть связано что-то с HA и MQTT?

Whilser, 2020-12-08:
> А вот как сделать в ХА будильник через MQTT? Мне очень интересно.

Пока такого функционала нет, но теоретически можно реализовать и тогда будильник можно будет устанавливать выполнив публикацию в ветку MQTT. Это можно будет делать из панели разработчика либо автоматизациями, даже можно настроить из карточки ХА при желании. Весь вопрос насколько такая функция будет востребована пользователями и насколько это будет удобно.

balbulator, 2020-12-08:
Удобно-удобно, потому как можно привязать к умному будильнику или к реальному рассвету, или к состоянию жалюзи на окне. Главное вытянуть в НА суть включения режима "рассвет". Также подумываю привязать цветовую палитру к температуре в доме или к погоде на улице. Еще "Полицейскую сирену" можно прикрутить к сигнализации. Идей уйма. 

Заметил, при загрузке страницы по IP лампы, лампа зависает, как только появляется сообщение "Страница недоступна", лампу попускает, ищу проблему

closed 2022-01-05 by Whilser