Whilser/GyverLamp #32: Интеграция с датчиком со2 и температуры

lexxnsk, 2021-01-12:
Есть ли возможность добавить работу с датчиками mh-z19b и bme280?

lexxnsk, 2021-01-12:
Могу попробовать сделать сам - подскажите куда копать.

Whilser, 2021-01-14:
Теоретически это возможно, но на практике это может привести к дополнительной нестабильности. Пример скетча подключения датчика углекислого газа можно найти в сети, например, здесь - https://arduino-technology.ru/coding/hardware/mhz19b-sensor/ Все что останется, это добавить запрос статуса с датчика в таймер (существующий либо создать свой). 

Наиболее оптимальным вариантом будет подключить эти датчики на отдельную ESP и залить туда прошивку ESPHome, Tasmota и т.п.

lexxnsk, 2021-01-14:
Спасибо за ответ.

closed 2022-01-05 by Whilser