Whilser/GyverLamp #34: First LED of Matrix is flashing/flickering green

Puh73, 2021-06-19:
I have the problem, that the first LED of my 16x16 LED Matrix is flashing/flickering green. I already have googled this problem and found out that I am not the only one with this problem. In Internet I found many people who have problems with NodeMCU, ESP8266, Wemos D1, and other boards and WS2812 LED-Strips with flashing/flickering green first LED-Pixel and no one have solved this issue. I have added an levelshifter from 3,3V to 5V on LED_PIN to LED-Matrix, but that don’t helped. I tried out the FastLED Demo with mostly the same effects and this working without any blinking or flicking first LED. Am I the only one with this flashing first LED-Pixel problem? Thank you in advance.

Patrick

tema77, 2021-07-17:
У меня такая же проблема. Помогите, пожалуйста, разобраться.

Puh73, 2021-07-18:
> У меня такая же проблема. Помогите, пожалуйста, разобраться.

It’s the ESP8266 Driver - you only have to downgrade to 2.7.4 - don’t use the 3.x ESP-8266 Driver

Puh73, 2021-07-18:
[image]

tema77, 2021-07-19:
Thank you so much!

closed 2022-01-05 by Whilser